m5stack/M5CoreS3 #2: Failing to compile due to missing file "bmi270.h"

tinytaro, 2023-05-08:
Run `pio run` command:

```
Archiving .pio\build\esp32-s3-devkitc-1\libd4a\libSPI.a
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\HardwareSerial.cpp.o
In file included from src/utility/I2C_IMU.cpp:1:
src/utility/I2C_IMU.h:7:10: fatal error: bmi270.h: No such file or directory

****************************************************************
* Looking for bmi270.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bmi270.h"
* Web  > https://registry.platformio.org/search?q=header:bmi270.h
*
****************************************************************

 #include "bmi270.h"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32-s3-devkitc-1\src\utility\I2C_IMU.cpp.o] Error 1
In file included from src/M5CoreS3.h:16,
                 from src/M5CoreS3.cpp:5:
src/utility/I2C_IMU.h:7:10: fatal error: bmi270.h: No such file or directory
```

IcingTomato, 2023-05-08:
OK, I will test it.

Tinyu-Zhao, 2023-05-08:
Temporary solution, follow the links to install both libraries manually

https://github.com/boschsensortec/BMI270-Sensor-API

https://github.com/BoschSensortec/BMM150-Sensor-API

tunefs, 2023-05-09:
I have installed these. Now I get another error.
```
/Users/saito/Documents/Arduino/libraries/M5CoreS3/src/utility/I2C_IMU.cpp: In member function 'void I2C_IMU::Update()':
/Users/saito/Documents/Arduino/libraries/M5CoreS3/src/utility/I2C_IMU.cpp:135:13: error: 'bmm150_aux_raw_mag_data' was not declared in this scope
             bmm150_aux_raw_mag_data(bmi_sensor_data.aux_data, &bmm_mag_data,
             ^~~~~~~~~~~~~~~~~~~~~~~
/Users/saito/Documents/Arduino/libraries/M5CoreS3/src/utility/I2C_IMU.cpp:135:13: note: suggested alternative: 'bmm150_aux_mag_data'
             bmm150_aux_raw_mag_data(bmi_sensor_data.aux_data, &bmm_mag_data,
             ^~~~~~~~~~~~~~~~~~~~~~~
             bmm150_aux_mag_data
```

wianoski, 2023-05-10:
got this error too 
[image]

3110, 2023-05-10:
The following code can be used to compile this library.
https://github.com/imliubo/CoreS3-UserDemo/tree/main/lib/BMM150-Sensor-API
https://github.com/imliubo/CoreS3-UserDemo/tree/main/lib/BMI270-Sensor-API

Tinyu-Zhao, 2023-05-10:
Please install this repository and try to compile again.
@3110 @tunefs @tinytaro @wianoski

tunefs, 2023-05-10:
I was able to build it successfully and it works on Core S3. Thank you very much.

closed 2023-05-10 by Tinyu-Zhao

3110, 2023-05-11:
@Tinyu-Zhao Could you add the following in `platformio.ini`?
``
lib_deps =
    https://github.com/Tinyu-Zhao/BMI270_Sensor
```

Tinyu-Zhao, 2023-05-12:
> The following code can be used to compile this library.
> https://github.com/imliubo/CoreS3-UserDemo/tree/main/lib/BMM150-Sensor-API
> https://github.com/imliubo/CoreS3-UserDemo/tree/main/lib/BMI270-Sensor-API

Added to platform.ini