m5stack/M5CoreS3 #3: uiflow?

d4rkdpg, 2023-05-08:
How I flash uiflow onto this, i just received a batch of them :). And I wanted to migrate my core2 projects to this?

imliubo, 2023-05-08:
Hi @d4rkdpg , cores3 uiflow support is not fully ready, and we develop new uiflow to support esp32-s3 chips device, you can check [here](https://uiflow2.m5stack.com/) but still in alpha version, cores3 support almost ready.

d4rkdpg, 2023-05-08:
how can I add to apha version, I dont see how?

imliubo, 2023-05-08:
> how can I add to alpha version, I dont see how? 

CoreS3 FW is not release yet.

> also alpha version looks to be missing a lot of stuff like remote+, lorawan intergration etc.

Yeah, uiflow 2.0 has big change, module and unit we are supporting now, we will support all module and unit ASAP.

d4rkdpg, 2023-05-08:
Can I asked when this will be? I need the asap for production.  also atoms3 lite I need all features of uiflow1.

d4rkdpg, 2023-06-06:
I haven't see much progress here thanks for uiflow 2 support, but we need some ui buttons, sliders, remote plus support, lora module support etc.  it's really unusable

pgsinoz, 2023-06-18:
will it include 8encoder module too?

pgsinoz, 2023-07-07:
> will it include 8encoder module too?

confirming this is now fixed in V2.00 Alpha 19 thanks

Tinyu-Zhao, 2024-06-07:
> > will it include 8encoder module too?它会包括 8 个编码器模块吗？
> 
> confirming this is now fixed in V2.00 Alpha 19 thanks确认这在 V2.00 Alpha 19 中已修复，谢谢

Closed.

closed 2024-06-07 by Tinyu-Zhao