m5stack/M5CoreS3 #10: Not working in arduino 1.8.9

wil-sys, 2023-05-14:
Even with all dependencies installed (M5stack boards/libraries, M5Family, etc), the CoreS3 is not shown on the boards list, and example code does not run even when the board "ESP32-s3 dev kit it selected"

felmue, 2023-05-14:
Hello @wil-sys 

according to the release notes Arduino IDE 1.8.9 is from 2019.

It looks like support for the S3 variant was only added in Arduino IDE 2.0.3. Please see [here](https://esp32s3.com/getting-started.html).

Quote from the article:

_Arduino support for the ESP32-S3 is available in the Espressif ESP32 Arduino Core **2.0.3** and later._

I suggest you upgrade your Arduino IDE and try again.

Thanks
Felix

wil-sys, 2023-05-14:
The board still doesn’t show up, and does not compile in arduino IDE 2.1.0.

LordLeXaT, 2023-05-14:
I confirm its compiling (with current lib) on arduino 1.8.9
[image]

1. The lib should be installed to Arduino/libraries (not board library)
2. Boardsetup like in the screenshoot
3. tinyu-zhao/BMI270_Sensor should also be in libraries folder

for me its compiling with no issues if 1 - 3 are done

Tinyu-Zhao, 2023-05-16:
You need to update the board definition of M5Stack to 2.0.7.

Specific tutorials [link](https://docs.m5stack.com/zh_CN/quick_start/arduino)

[image]

[image]

LordLeXaT, 2023-05-16:
> You need to update the board definition of M5Stack to 2.0.7.
> 
> Specific tutorials [link](https://docs.m5stack.com/zh_CN/quick_start/arduino)

I think  you got something wrong :) I have no Issues :D I just tried to help