m5stack/M5CoreS3 #16: Basics/Camera example building issue

lboue, 2023-05-26:
Hello,

I am facing an issue building [Basics/Camera](https://github.com/m5stack/M5CoreS3/blob/main/examples/Basics/Camera/Camera.ino
) example.
```
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2023426-33536-thpr5h.d7lcs\Camera\Camera.ino:4:10: fatal error: CoreS3_GC0308.hpp: No such file or directory
 #include <CoreS3_GC0308.hpp>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: CoreS3_GC0308.hpp: No such file or directory
```

Do you know how to fix this issue?

### Screenshot

[image]

LordLeXaT, 2023-05-28:
@lboue 
in the INO File change
#include <CoreS3_GC0308.hpp>
to
#include "CoreS3_GC0308.hpp"

should now include from the sketch folder, not from a library and should compile

lboue, 2023-05-29:
Thank you, this solved my compilation issue.

Do you know why I have this warning?
```
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2023429-23212-11xmyrw.r9i6\Camera\app_httpd.cpp:17:18: note: #pragma message: Disable face detection and recognition
 # pragma message "Disable face detection and recognition"
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

closed 2023-05-29 by lboue