m5stack/M5CoreS3 #20: Basics/Camera example cam_hal: cam_dma_config(306): frame buffer malloc failedESP-ROM:esp32s3-20210327

lboue, 2023-05-30:
Hello,

I am facing when running the [Basics/Camera](https://github.com/m5stack/M5CoreS3/blob/main/examples/Basics/Camera/Camera.ino
) example.

### Setup

- Arduino IDE 2.1.0
- W11

### Full serial log:
```
07:37:27.890 -> E (462) cam_hal: cam_dma_config(306): frame buffer malloc failedESP-ROM:esp32s3-20210327
07:37:27.890 -> Build:Mar 27 2021
07:37:27.890 -> rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
07:37:27.890 -> Saved PC:0x420929d6
07:37:27.890 -> SPIWP:0xee
07:37:27.890 -> mode:DIO, clock div:1
07:37:27.890 -> load:0x3fce3808,len:0x44c
07:37:27.890 -> load:0x403c9700,len:0xbec
07:37:27.890 -> load:0x403cc700,len:0x2920
07:37:27.890 -> entry 0x403c98d8
07:37:28.025 -> E (181) esp_core_dump_flash: No core dump partition found!
07:37:28.025 -> E (181) esp_core_dump_flash: No core dump partition found!
07:37:28.476 -> E (460) cam_hal: cam_dma_config(306): frame buffer malloc failed
07:37:28.476 -> E (461) cam_hal: cam_config(390): cam_dma_config failed
07:37:28.476 -> E (461) camera: Camera config failed with error 0xffffffff
```

LordLeXaT, 2023-05-30:
did you enable PSRAM?
[image]
otherwise it will not work, as per defined in code "config.fb_location = CAMERA_FB_IN_PSRAM;" the framebuffer is 
located in the PSRAM

lboue, 2023-05-30:
> did you enable PSRAM? [image] otherwise it will not work, as per defined in code "config.fb_location = CAMERA_FB_IN_PSRAM;" the framebuffer is located in the PSRAM.

I was not enabled by default. In fact, I hadn't paid enough attention.
Perhaps it should be indicated in the code to avoid oversights.

> PS: I told you this already in the last issue you opened.
> 
> PPS: before you ask this, If you compile the code, the cam image looks strange, but there is already a pull request ongoing to fix this #19 by @felmue , If you look into the code changes you can see what you may need to change

Thanks. It works fine after this:
[image]

closed 2023-05-30 by lboue