m5stack/M5CoreS3 #21: Compilation error: 'USBSerial' was not declared in this scope

lboue, 2023-05-30:
Hello,

It looks like a line is missing to declare the variable in [HelloWorld.ino](https://github.com/m5stack/M5CoreS3/blob/main/examples/Basics/HelloWorld/HelloWorld.ino) example: 

```
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2023430-5812-sfr3mr.33j4\HelloWorld\HelloWorld.ino:32:5: error: 'USBSerial' was not declared in this scope
     USBSerial.println(
     ^~~~~~~~~
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2023430-5812-sfr3mr.33j4\HelloWorld\HelloWorld.ino:32:5: note: suggested alternative: 'Serial'
     USBSerial.println(
     ^~~~~~~~~
     Serial

exit status 1

Compilation error: 'USBSerial' was not declared in this scope
```

felmue, 2023-05-31:
Hello @lboue 

have you selected the correct board? One which is an **S3** variant of the ESP32?

Thanks
Felix

lboue, 2023-05-31:
> Hello @lboue
> 
> have you selected the correct board? One which is an **S3** variant of the ESP32?
> 
> Thanks Felix

Yes I did. I selected "M5Stack-CoreS3.

[image]

lboue, 2023-05-31:
Here is the menu:
[image]

felmue, 2023-05-31:
Hello @lboue 

hmm, ok. Then I am afraid I don't know why it fails to compile. Sorry.

Thanks
Felix

felmue, 2023-05-31:
Hello @lboue 

hmm ok. I am afraid I don't know why the compilation fails then. Sorry.

Thanks
Felix

lboue, 2023-05-31:
It must be is an issue with the code because serial port works fine with this example code with no includes:
```
void setup() {
  Serial.begin(9600);

  Serial.println("SERIAL");
  delay(500);
}

void loop() {
  // basic readout test, just print the current temp
  
  Serial.print("SERIAL"); 
  delay(1000);
}
```

[image]

lboue, 2023-05-31:
This example works when I replace USBSerial by Serial in the included files. See my PR.

Tinyu-Zhao, 2023-06-04:
You need to select this.
[image]

closed 2023-06-07 by Tinyu-Zhao