m5stack/M5CoreS3 #22: Fail to build I2C_Tester.ino example:  error: 'USBSerial' was not declared in this scope

lboue, 2023-05-30:
### Summary

It fails to build [I2C_Tester.ino](https://github.com/m5stack/M5CoreS3/blob/main/examples/Advanced/I2C_Tester/I2C_Tester.ino) example

### Error
It seems to have problem with the management of the serial port.
```
src\M5CoreS3.cpp:21:5: error: 'USBSerial' was not declared in this scope
     USBSerial.printf("APX initial:%d\n", Axp.begin(&Wire1));
```

### Arduino IDE Logs
```
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\utility\I2C_IMU.cpp: In constructor 'I2C_IMU::I2C_IMU()':
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\utility\I2C_IMU.cpp:10:5: error: 'USBSerial' was not declared in this scope
     USBSerial.begin(115200);
     ^~~~~~~~~
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\utility\I2C_IMU.cpp:10:5: note: suggested alternative: 'Serial'
     USBSerial.begin(115200);
     ^~~~~~~~~
     Serial
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\utility\I2C_IMU.cpp: In member function 'void I2C_IMU::Init()':
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\utility\I2C_IMU.cpp:96:5: error: 'USBSerial' was not declared in this scope
     USBSerial.printf("Valid BMM150 (Aux) sensor - Chip ID : 0x%x\r\n",
     ^~~~~~~~~
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\utility\I2C_IMU.cpp:96:5: note: suggested alternative: 'Serial'
     USBSerial.printf("Valid BMM150 (Aux) sensor - Chip ID : 0x%x\r\n",
     ^~~~~~~~~
     Serial
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\M5CoreS3.cpp: In member function 'void M5CoreS3::begin(bool, bool, bool)':
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\M5CoreS3.cpp:15:9: error: 'USBSerial' was not declared in this scope
         USBSerial.begin(115200);
         ^~~~~~~~~
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\M5CoreS3.cpp:15:9: note: suggested alternative: 'Serial'
         USBSerial.begin(115200);
         ^~~~~~~~~
         Serial
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\M5CoreS3.cpp:21:5: error: 'USBSerial' was not declared in this scope
     USBSerial.printf("APX initial:%d\n", Axp.begin(&Wire1));
     ^~~~~~~~~
c:\Users\user\Documents\Arduino\libraries\M5CoreS3\src\M5CoreS3.cpp:21:5: note: suggested alternative: 'Serial'
     USBSerial.printf("APX initial:%d\n", Axp.begin(&Wire1));
     ^~~~~~~~~
     Serial

exit status 1

Compilation error: exit status 1
```

lboue, 2023-05-31:
[image]

Tinyu-Zhao, 2023-06-04:
Similar questions #21

closed 2023-06-04 by Tinyu-Zhao