m5stack/M5CoreS3 #25: [Request] Add Basics/speak example for CoreS3

lboue, 2023-06-01:
Hello,

Could you add this [Basics/speak](https://github.com/m5stack/M5Core2/blob/master/examples/Basics/speak/speak.ino) example's equivalent of the Core2 card for the CoreS3 model?

Regards

IcingTomato, 2023-06-08:
@Tinyu-Zhao

YonatanAmir1996, 2023-08-25:
Hi @IcingTomato, I hope you're doing well. It's been two months since we last discussed the support feature. Would you be able to add it? Thanks for considering!

IcingTomato, 2023-08-25:
> Hi @IcingTomato, I hope you're doing well. It's been two months since we last discussed the support feature. Would you be able to add it? Thanks for considering!

[image]

YonatanAmir1996, 2023-08-26:
> Hi @IcingTomato, I hope you're doing well. It's been two months since we last discussed the support feature. Would you be able to add it? Thanks for considering!

@Tinyu-Zhao

Tinyu-Zhao, 2023-09-04:
You can try this program, the function on line 72 is used to generate the audio, you can also replace the audio with another one.
[cores3-speaker-master.zip](https://github.com/m5stack/M5CoreS3/files/12515124/cores3-speaker-master.zip)

closed 2023-09-04 by Tinyu-Zhao