m5stack/M5CoreS3 #26: [Request] Add Factory Test code to the repo

markwjkelly, 2023-06-06:
Are you planning on adding the Factory Test code to the Repo.  That has always been a great addition to all your dev kits, especially as it helps developers get a better understanding of the complete system.

luisllamasbinaburo, 2023-06-06:
I restore Factory Test and create a PR.
While we wait for it to be accepted, you can have the code here https://github.com/luisllamasbinaburo/M5CoreS3

Tinyu-Zhao, 2023-06-07:
> Are you planning on adding the Factory Test code to the Repo. That has always been a great addition to all your dev kits, especially as it helps developers get a better understanding of the complete system.

Factory test program is too large, will cause the download time to extend many times, you can refer to this repository.

closed 2023-06-07 by Tinyu-Zhao

luisllamasbinaburo, 2023-06-07:
Maybe you can create a different repo for M5StackCoreS3 Factory Demo?
Like you do, for example, for de M5Paper? https://github.com/m5stack/M5Paper_FactoryTest

Tinyu-Zhao, 2023-06-07:
> Maybe you can create a different repo for M5StackCoreS3 Factory Demo?
> 
> Like you do, for example, for M5Paper? https://github.com/m5stack/M5Paper_FactoryTest

Ok，I will.