m5stack/M5CoreS3 #30: Pls detail arduino 2.1.0 Tools config settings to build camera.ino

achalmers, 2023-06-06:
Have been ableto build all examples except Camera (required updating arduino IDE and many dependent libraries). Spent several days on the Camera.ino playing with Tools options in Arduino2.1.0 but still getting weird serial port failure errors reported at 75% programmed .
I noticed the partitions.csv is included in the Camera directory. Im guessing it gets used automatically,  im wondering if it is correct also ? (Ive seen errors "No core dump partition found!")  Heres the original csv.
[partitions.csv](https://github.com/m5stack/M5CoreS3/files/11667543/partitions.csv)

Also noticed no sample code for touchscreen, or the very nice factory application.

achalmers, 2023-06-06:
Applying the 2 new firmware updates fixed the serial port error on download (I also slowed down burningto115200). I suspect the csv needs a line for a coredump partition to be added in order to debug the (last?) runtime issue. I also see the FactoryApp code was added to the repo for PlatformIO.

IcingTomato, 2023-06-08:
Would you like to provide more information about camera demo serial output message on Arduino 2.1.0?

achalmers, 2023-06-08:
Before applying the 2 firmware updates (see https://docs.m5stack.com/en/core/CoreS3 ) the Camera.ino would compile but during the program stage it would stop at 75% complete and generate a 'serial port error'.

After applyingthose hotfixes/firmare updates I now am able to build the Camera.ino and program the device, but when it starts execution the serial output shows :   
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42092d76
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x27c0
entry 0x403c98c0
E (206) esp_core_dump_flash: No core dump partition found!
E (206) esp_core_dump_flash: No core dump partition found!

Im guessing I havea Tool/Setting incorrect as well as no dump partition in the partitions_ffat.csv   .

IcingTomato, 2023-06-10:
[image]
Would you like to open PSRAM and try one more time?

IcingTomato, 2023-06-10:
I uploaded a PlatformIO demo just now. 
https://github.com/m5stack/M5CoreS3/tree/6e0c70bd913a4a8bc98fcc861652cb3c5aba882b
You can have a try.

achalmers, 2023-06-10:
Yes PSRAM was set see Settings below.   Serial Output also pasted
[image]

[image]

IcingTomato, 2023-06-11:
Strangely enough, could you try the PlatformIO camera example?
https://github.com/m5stack/M5CoreS3/tree/main/examples/PlatformIO/Camera
I cannot reproduce your mistakes. Would you like to erase all the flash and re-upload the codes?
[image]

achalmers, 2023-06-12:
Yes I did erase flash.

Thx for adding the Platformio project. 
OK, Ive been holding off updating to Platformio but I guess its time.
I downloaded the latest Platformio and loaded camera project and required libraries, the camera.ini build seems to halt when freeRTOS (called by esp_camera ?)  needs a setting, see below for error. Not sure what to set or how to define (eg add --device xxxx  to its line in the library section ?). Probably due tome not using Platformio before. 

[image]

IcingTomato, 2023-06-14:
Oh, I see.
You should pick the demo document out of the library.
[image]