m5stack/M5CoreS3 #36: [Request] Power Management example code

webstep-pl, 2023-07-02:
In the URL https://docs.m5stack.com/en/core/CoreS3 there is a graphic representing Power Management options:
USB IN, BUS IN => BUS_OUT_EN = 0, USB_OTG_EN = 0
USB IN, BUS OUT => BUS_OUT_EN = 1, USB_OTG_EN = 0
USB OUT, BUS IN => BUS_OUT_EN = 0, USB_OTG_EN = 1
USB IN, BUS OUT => BUS_OUT_EN = 1, USB_OTG_EN = 1

[image]

I would like to control it within the app, but there is no factory test released for CoreS3 which I would look into + no example provided for that.

Apart from publishing an example to the community, could you please also share an inline code how to make it?
M5.Axp._something here_

I have tested many things by looking into AXP2101.h and reusing public methods, but no luck so far.
Help would be very appreciated!

Tinyu-Zhao, 2023-08-24:
Now we have power control functions added to the AXP2101.
https://github.com/m5stack/M5CoreS3/blob/fdbb037bf9ea65569542c37be23607ecc2a57cd8/src/AXP2101.h#L107-L111

d4rkdpg, 2024-05-10:
Hi can you please tell me how to call powerModeSet, I have a DMX module and when I used 12v in the display wont turn on, keeps power cycling. thanks

Tinyu-Zhao, 2024-06-04:
> Hi can you please tell me how to call powerModeSet, I have a DMX module and when I used 12v in the display wont turn on, keeps power cycling. thanks

eg: powerModeSet(mode)

mode:
    POWER_MODE_USB_IN_BUS_IN = 0,
    POWER_MODE_USB_IN_BUS_OUT,
    POWER_MODE_USB_OUT_BUS_IN,
    POWER_MODE_USB_OUT_BUS_OUT,

lidense, 2025-02-05:
Hi I would like to use my M5stackCoreS3 with TFT_eSPI, without using the CoreS3 and Unified libraries. 
I understand I have to switch on and power the TFT screen with some command.
Please advise.

lidense, 2025-02-05:
When I run tft.init() I get a crash

Tinyu-Zhao, 2025-02-06:
> When I run tft.init() I get a crash

You can refer to this branch.

https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7

lidense, 2025-02-06:
OK but how do I integrate TFT_eSPI into this?
Regards

----- Mail original -----

https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7

Tinyu-Zhao, 2025-02-06:
Isn't this TFT_eSPI?

https://github.com/m5stack/M5CoreS3/blob/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility/In_eSPI.h
 
https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility

Tinyu-Zhao, 2025-02-06:
> OK but how do I integrate TFT_eSPI into this?
> Regards
> […](#)

Isn't this TFT_eSPI?

https://github.com/m5stack/M5CoreS3/blob/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility/In_eSPI.h
 
https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility

lidense, 2025-02-06:
OK, I'll have a close look into this.
Thanks a lot for your prompte answer, much appreciated.
Regards
Dave

----- Mail original -----

https://github.com/m5stack/M5CoreS3/blob/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility/In_eSPI.h 

https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility

lidense, 2025-02-06:
Im trying to use this library to which I have added another library (ESP32-audioI2S) but when I compile I get errors such as:
.pio/libdeps/M5CoreS3/ESP32-audioI2S/src/Audio.h:17:10: fatal error: WiFiClientSecure.h: No such file or directory
while this  WiFiClientSecure.h does exist and I did not have this kind of error with the common M5Core3 library.
Please advise.
Thanks

----- Mail original -----

https://github.com/m5stack/M5CoreS3/blob/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility/In_eSPI.h 

https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility

lidense, 2025-02-06:
...for information I added M5Unified because one of the example files included it...

----- Mail original -----

https://github.com/m5stack/M5CoreS3/blob/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility/In_eSPI.h 

https://github.com/m5stack/M5CoreS3/tree/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7/src/utility