m5stack/M5CoreS3 #37: Linking issue on PlatformIO build

hpsaturn, 2023-07-08:
## Overview

After clone of the repo, and then perform a `pio run` I have the next linking issue:

```cpp
Linking .pio/build/M5CoreS3/firmware.elf
/home/avp/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/M5CoreS3/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
/home/avp/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/M5CoreS3/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
/home/avp/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/M5CoreS3/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':
/home/avp/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42: undefined reference to `setup()'
/home/avp/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/avp/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:48: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
*** [.pio/build/M5CoreS3/firmware.elf] Error 1
```

Tinyu-Zhao, 2023-07-10:
Please attach the program you executed.

Tinyu-Zhao, 2023-07-10:
> yes, I think that the main platformio.ini of any library should have some examples or tests like a targets. For example in my library [CanAirIO Sensors lib](https://github.com/kike-canaries/canairio_sensorlib), this library has the samples like unit tests in the main platformio.ini file. Maybe also you should include the Factory Sample like a target. Right now this source code is missing here and with that it would be nice to understand better this demo and also test this library.

I may understand what you mean, but we are not trying to transform this repository into a platformio project, the ini file is just a configuration file for the platformio project compilation of cores3

hpsaturn, 2023-07-10:
well, the idea is not tranform it to a PlatformIO project, the idea could be testing the library or its examples. On the other hand I'm going to write a new issue, to request the Factory source code for the CoreS3, because I can't find it. Thanks in advance if you know were is it :)

Tinyu-Zhao, 2023-07-21:
> yes, I think that the main platformio.ini of any library should have some examples or tests like a targets. For example in my library [CanAirIO Sensors [...]

You can submit a simple PR with your idea to see if it will be accepted.

closed 2023-07-21 by Tinyu-Zhao

hpsaturn, 2023-07-24:
Done. Is ready a new PR to resolve this issue