m5stack/M5CoreS3 #39: PaHub(U040-B) and PbHub(U041-B) support

YonatanAmir1996, 2023-07-18:
Add support for PaHub(U040-B) and PbHub(U041-B)

Tinyu-Zhao, 2023-07-21:
You can try this.
[PaHUB.zip](https://github.com/m5stack/M5CoreS3/files/12116190/PaHUB.zip)

YonatanAmir1996, 2023-07-21:
@Tinyu-Zhao 
1. PaHub for CoreS3 following code worked for me:
https://github.com/m5stack/M5-ProductExampleCodes/blob/master/Unit/PaHUB/Arduino/PaHUB/PaHUB.ino 
2.Tested Heart unit rate in CoreS3 - Works.
3.Issues:
. But there are issues with U003(RGB Unit) + N20(Vibration motor) with official example code. Tried to change Pin to - 8 (Like schematic) but didn't work and there isn't example of how can  wire those port B to PbHub.

Tinyu-Zhao, 2023-09-04:
The reason may be that the PortA power is not turned on, you can use M5.Axp.powerModeSet(POWER_MODE_USB_IN_BUS_OUT) to turn it on.

closed 2023-09-04 by Tinyu-Zhao