m5stack/M5CoreS3 #45: WIFI basic HTTP Client  #if with no expression

sallah21, 2023-08-26:
I got got compilation error for WIFI basic HTTPClient example
Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32c3/include/hal/include/hal/i2c_types.h:85:28: error: #if with no expression
 #if CONFIG_IDF_TARGET_ESP32

sallah21, 2023-08-28:
examples/Advanced/WIFI/BasicHttpClient/BasicHttpClient.ino

Tinyu-Zhao, 2023-08-28:
[image]
The following compilation configuration failed to reproduce, please specify the compilation environment you are using and the libraries and board definitions you are using.
[env:M5CoreS3]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
upload_port = /dev/cu.usbmodem144101
upload_speed = 1500000
monitor_speed = 115200
monitor_port = /dev/cu.usbmodem144101
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -DCORE_DEBUG_LEVEL=4

Resonious, 2023-09-04:
I'm getting this too, with just the Core S3 "hello world" program:

```c++
#include <M5CoreS3.h>

void setup() {
    M5.begin(true, true, false); 
    M5.Lcd.setTextSize(2);
    M5.Lcd.setCursor(100, 120);
    M5.Lcd.print("Hello World");
}

void loop() {
    USBSerial.println("Hello World");
    delay(500);
}
```

Tinyu-Zhao, 2023-09-04:
> I'm getting this too, with just the Core S3 "hello world" program:
>
>
>
> ```c++
>
> #include <M5CoreS3.h>
>
>
>
> void setup() {
>
>     M5.begin(true, true, false);
>
>     M5.Lcd.setTextSize(2); [...]

Please post the program or the address of the program

Resonious, 2023-09-05:
Sorry, probably a rookie mistake. I just grabbed `arduino-cli` and ran

```
arduino-cli core install esp32:esp32
arduino-cli lib install M5CoreS3
```

My `~/.arduino15/arduino-cli.yaml` looks like

```
board_manager:
  additional_urls: 
    - https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json
build_cache:
  compilations_before_purge: 10
  ttl: 720h0m0s
daemon:
  port: "50051"
directories:
  data: /home/nigel/.arduino15
  downloads: /home/nigel/.arduino15/staging
  user: /home/nigel/Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: false
updater:
  enable_notification: true
```

and I've run `arduino-cli core update-index`. I'm guessing there's something else I need to do, but I'm not sure what it is.

Resonious, 2023-09-05:
May I ask how?

Tinyu-Zhao, 2023-09-05:
> May I ask how?

https://dev.to/stepanvrany/esp32-with-arduino-cli-36mh

Arduino config file arduino-cli.yaml Replace the link in https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json

closed 2023-09-05 by Tinyu-Zhao

Resonious, 2023-09-05:
Thanks for the link! I've already followed those steps (you can see the `package_m5stack_index.json` in my config file in my earlier message). Here's some full output:

```
[nigel@reswork Watch]$ arduino-cli core update-index
Downloading index: package_index.tar.bz2 downloaded                                                                                         
Downloading index: package_m5stack_index.json downloaded                                                                                    
[nigel@reswork Watch]$ arduino-cli core install esp32:esp32
Platform esp32:esp32@2.0.11 already installed
[nigel@reswork Watch]$ arduino-cli board list
シリアルポート      Protocol タイプ               Board Name         FQBN                Core       
/dev/ttyACM0 serial   Serial Port (USB) ESP32C3 Dev Module esp32:esp32:esp32c3 esp32:esp32

[nigel@reswork Watch]$ arduino-cli compile -b esp32:esp32:esp32c3 .
In file included from /home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32c3/include/driver/include/driver/i2c.h:24,
                 from /home/nigel/Arduino/libraries/M5CoreS3/src/utility/I2C_IMU.h:5,
                 from /home/nigel/Arduino/libraries/M5CoreS3/src/M5CoreS3.h:16,
                 from /home/nigel/p/Watch/Watch.ino:1:
/home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32c3/include/hal/include/hal/i2c_types.h:85:28: error: #if with no expression
 #if CONFIG_IDF_TARGET_ESP32
                            ^

Used library  Version Path                                                                      
M5CoreS3      0.0.4   /home/nigel/Arduino/libraries/M5CoreS3                                    
Wire          2.0.0   /home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/Wire
FS            2.0.0   /home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/FS  
SD            2.0.0   /home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/SD  
SPI           2.0.0   /home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/SPI 
BMI270_Sensor 0.0.1   /home/nigel/Arduino/libraries/BMI270_Sensor                               

Used platform Version Path                                                       
esp32:esp32   2.0.11  /home/nigel/.arduino15/packages/esp32/hardware/esp32/2.0.11
Error during build: exit status 1
```

> So why don't you use an IDE

Personal preference :grin:. The doc you linked to has a CLI section too, though without as much info.

I'll try with the IDE and see what happens.

Tinyu-Zhao, 2023-09-05:
> Thanks for the link! I've already followed those steps (you can see the `package_m5stack_index.json` in my config file in my earlier message). Here's some full output:
>
>
>
> ```
>
> [nigel@reswork Watch]$ [...]

I don't know why when you use the arduino-cli board list command you get the FQBN as esp32c3 and the M5CoreS3 chip as esp32s3, can you be sure you're using the M5CoreS3 host?

Try this: arduino-cli compile -b esp32:esp32:esp32s3

Resonious, 2023-09-05:
Ah! Your compile command worked. Upload also worked :+1: thanks a bunch @Tinyu-Zhao