m5stack/M5CoreS3 #46: SD card example fails after reset

felmue, 2023-08-27:
Hello guys
for me the SD card example only works once after powering M5CoreS3 up, but not after pressing the reset button.

Below is the debug output after pressing the reset button:
```
M5CoreS3 initializing...OK
[  2492][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
APX initial:0
[  2496][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  2863][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=2 scl=1 freq=100000
[  2886][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2987][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3087][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3187][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  3187][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  3188][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3197][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3302][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3403][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3504][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card failed, or not present
```
As reference the log output after powering up M5CoreS3:
```
M5CoreS3 initializing...OK
[  2492][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
APX initial:0
[  2496][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  2863][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=2 scl=1 freq=100000
TF card initialized.
hello.txt exists.
Creating hello.txt
Writing to test.txt...
done.
/hello.txt Content:
SD test.
```
I think it has to do how the LCD and SD card are sharing the SPI bus. Especially how MISO is used as D/C for the LCD.

Thanks
Felix

Tinyu-Zhao, 2023-09-12:
> Hello guys
>
> for me the SD card example only works once after powering M5CoreS3 up, but not after pressing the reset button.
>
>
>
> Below is the debug output after pressing the reset button:
>
> ``` [...]

There is a spi conflict problem, the program closes the communication between spi and lcd after executing the lcd operation just to prevent the conflict, but I have reproduced it according to your method, and there is no problem as you said, no matter how many times I restarted, I can operate the sdcard normally.

felmue, 2023-09-12:
Hello @Tinyu-Zhao 

thank you for testing on your side.

When you say 'restarted' did you actually press the reset button on M5CoreS3?

BTW: when I comment out the second print to the LCD it works for me after a reset as well, e.g.
```
//    M5.Lcd.print("Pls open the serial port.");
```
I guess there is a subtle timing issue?

Thanks
Felix

Tinyu-Zhao, 2023-09-14:
> Hello @Tinyu-Zhao
>
> thank you for testing on your side.
>
> When you say 'restarted' did you actually press the reset button on M5CoreS3?
>
> BTW: when I comment out the second print to the LCD it works [...]

https://github.com/m5stack/M5CoreS3/assets/44396641/63e1c19a-3eb3-4601-8f0a-d3697b691c3b

felmue, 2023-09-15:
Hello @Tinyu-Zhao 

thank you for the movie - unfortunately I cannot see the picture (probably an issue with Ubuntu and / or Firefox). But from the clicking sound I gather you pressed the reset button.

Anyway, thank you for your time - I am still puzzled though as to why it fails on my M5CoreS3.

Thanks
Felix

kjw140473, 2023-11-04:
Hi 

I am also having difficulty with this example.  I am trying to write to the LCD in the loop function.

But it does not seem like the SD library is releasing the SPI bus.

I have tried manually pulling the SDCard CS pin high at the end of the setup function.

Any help greatly appreciated

Gitshaoxiang, 2024-01-10:
add [sdcard example](https://github.com/m5stack/M5CoreS3/tree/main/examples/Basic/sdcard)

closed 2024-01-10 by Gitshaoxiang