m5stack/M5CoreS3 #49: Touch example .ino

jaylfc, 2023-09-13:
Can we get an example multifunction sketch showing us how to use screen/touch/sd card etc simultaneously. I'm struggling to get the features I need working together with the limited resources.

felmue, 2023-09-17:
Hello @jaylfc 

have you checked out the Touch [example](https://github.com/m5stack/M5Unified/blob/master/examples/Basic/Touch/Touch.ino) in the Unified repository? It shows how to use screen and touch. For the SD card maybe check out this [example](https://github.com/m5stack/M5Unified/blob/master/examples/Advanced/Speaker_SD_wav_file/Speaker_SD_wav_file.ino).

Thanks
Felix

jaylfc, 2023-09-19:
I have thanks ,they helped a little!

closed 2024-01-10 by Gitshaoxiang