m5stack/M5CoreS3 #50: Compilation error: 'USBSerial' was not declared in this scope

ropg, 2023-11-03:
I use the Arduino IDE (v.2.2.1) and have M5Stack board manager 2.0.8 and M5CoreS3 library v0.0.4

When I try to compile HelloWorld.ino from the examples directory, I get:
    Compilation error: 'USBSerial' was not declared in this scope

[image]

When I comment out the SerialUSB statement in the example, the compiler now complains about USBSerial in the I2C_IMU library.

[image]

&nbsp;

None of examples seem to compile, so I must be doing something wrong.

ylabo0717, 2023-11-15:
@ropg I also encountered the same problem, but I was able to resolve it by uninstalling V0.0.4 and installing the latest revision.

The following revision appears to have fixed the problem.
https://github.com/m5stack/M5CoreS3/commit/a9187b74aa3d6c90cc2cfc8f7e6d466ae3dd6cd7

Below is the procedure I have tried. I hope this will be helpful.

## 1. Uninstall V0.0.4

Remove V0.04 in Library Manager.

[image]

## 2. Download zip

Download the latest revision from this repository.

[image]

## 3. Install the zip to Arduino IDE

Select menu. Sketch -> Include Library -> Add .ZIP Library

[image]

Select the ZIP file you downloaded in Step 2.

[image]

## 4. Build

[image]

richstokes, 2023-11-19:
Thank you! this fixed it for me, too. The one installed via the package manager is broken.

ropg, 2023-11-20:
OK, some stuff now works. Sadly no example how to talk to the touch screen, some examples do not compile and boy these people hate documentation and clarity.

Tinyu-Zhao, 2024-06-07:
> OK, some stuff now works. Sadly no example how to talk to the touch screen, some examples do not compile and boy these people hate documentation and clarity.好的，现在一些东西可以运行。可惜没有关于如何与触摸屏交互的示例，有些示例无法编译，这些人讨厌文档和清晰度。

Add it.
https://github.com/m5stack/M5CoreS3/blob/9e73b0a61e91e7270850a0ca56c041f58cc108d3/examples/Basic/button/button.ino

closed 2024-06-07 by Tinyu-Zhao