m5stack/M5CoreS3 #52: MAX30010 issues when connected to PaHub

YonatanAmir1996, 2023-12-15:
Hi, In case of multiple I2C devices which one of them is MAX30010, making issues with MAX30010 device, such that the results are invalid, note that the I2C addresses between devices are different.

felmue, 2023-12-25:
Hello @YonatanAmir1996 

what are the I2C addresses of all the I2C devices you have connected? Note: If they are all different, then there is no need for a PaHub - you could simple connect them all directly to the core.

Thanks
Felix

YonatanAmir1996, 2023-12-30:
There is one I2C device which has the same address, but anyway, physically the max30100 can't be connected in a middle of row. Is there an alternative or changes needed to be done in MAX30100 API code?

felmue, 2024-01-01:
Hello @YonatanAmir1996 

I do not have a max30100 - so I cannot try myself whether it should work through PaHub or not. Sorry.

Thanks
Felix

Tinyu-Zhao, 2024-06-07:
Are you using Unit Heart?