m5stack/M5CoreS3 #54: M5ScreS3 - touch library issues

cohen-sydney, 2024-01-18:
### Describe the bug

I have CoreS3 and with the Arduino IDE V2 - I #include <M5CoreS3.h>
but the following  errors (on isPressed and the x & y error as well. 
I see the M5core2 API has these "functions" but I have CoreS3.

 if (M5.Touch.ispressed()) {
    int16_t x = M5.Touch.x;
    int16_t y = M5.Touch.y;

    if (x >= 285 && x <= 315 && y >= 25 && y <= 55) {
      connectToWiFi();
    }

### To reproduce

ARDUINO V2
M5Stack Core S3

### Expected behavior

no error

### Screenshots

[image]

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

cohen-sydney, 2024-01-19:
in following the definitions trail in the Ice, I can't see a Touch class relevant to the M%CoreS3, it references the Core2 touch call but this does not have button etc for the CoreS3??
is there a .h file for cores3 touch??

Lifeguard-alex, 2024-02-02:
yes its not working

Tinyu-Zhao, 2024-06-07:
You can try it.
```cpp
/**
 * @file button.ino
 * @author tinyu (tinyu@m5stack.com)
 * @brief M5CoreS3 Touch Test
 * @version 0.1
 * @date 2024-06-07
 *
 *
 * @Hardwares: M5CoreS3
 * @Platform Version: Arduino M5Stack Board Manager v2.0.9
 * @Dependent Library:
 * M5GFX: https://github.com/m5stack/M5GFX
 * M5Unified: https://github.com/m5stack/M5Unified
 * M5CoreS3: https://github.com/m5stack/M5CoreS3
 */

#include <M5CoreS3.h>

static m5::touch_state_t prev_state;

void setup(void) {
    auto cfg = M5.config();
    CoreS3.begin(cfg);

    CoreS3.Display.setTextColor(RED);
    CoreS3.Display.setTextDatum(middle_center);
    CoreS3.Display.setFont(&fonts::Orbitron_Light_24);
    CoreS3.Display.setTextSize(1);

    CoreS3.Display.drawString("Touch Button Test", CoreS3.Display.width() / 2, 15);
    CoreS3.Display.fillRect(0, CoreS3.Display.height() - 40, CoreS3.Display.width() / 3,
                            40, WHITE);
    CoreS3.Display.fillRect(CoreS3.Display.width() / 3, CoreS3.Display.height() - 40,
                            CoreS3.Display.width() / 3, 40, GREEN);
    CoreS3.Display.fillRect((CoreS3.Display.width() / 3) * 2,
                            CoreS3.Display.height() - 40, CoreS3.Display.width() / 3, 40,
                            YELLOW);
    CoreS3.Display.drawString("Btn A    Btn B    Btn C", CoreS3.Display.width() / 2,
                              CoreS3.Display.height() - 20);
}

void loop(void) {
    CoreS3.update();
    auto touchPoint = CoreS3.Touch.getDetail();
    if (prev_state != touchPoint.state) {
        prev_state = touchPoint.state;
    }
    if ((CoreS3.Display.height() > touchPoint.y &&
         touchPoint.y > CoreS3.Display.height() - 40) &&
        touchPoint.state == m5::touch_state_t::touch_begin) {
        CoreS3.Display.fillRect(0, 40, CoreS3.Display.width(), 70, BLACK);
        if (CoreS3.Display.width() / 3 > touchPoint.x && touchPoint.x > 0)
            CoreS3.Display.drawString("Btn A", CoreS3.Display.width() / 2,
                                      CoreS3.Display.height() / 2 - 30);
        if ((CoreS3.Display.width() / 3) * 2 > touchPoint.x &&
            touchPoint.x > CoreS3.Display.width() / 3)
            CoreS3.Display.drawString("Btn B", CoreS3.Display.width() / 2,
                                      CoreS3.Display.height() / 2 - 30);
        if (CoreS3.Display.width() > touchPoint.x &&
            touchPoint.x > (CoreS3.Display.width() / 3) * 2)
            CoreS3.Display.drawString("Btn C", CoreS3.Display.width() / 2,
                                      CoreS3.Display.height() / 2 - 30);
        Serial.printf("x:%d ,y:%d", touchPoint.x, touchPoint.y);
    }
}
```

closed 2024-06-07 by Tinyu-Zhao