m5stack/M5CoreS3 #56: M%CoreS3 & COM.X LTE Module

cohen-sydney, 2024-01-25:
### Describe the bug

is there any Examples of the COM.X LTE working with the M5CoreS3??

### To reproduce

using the Expmaple from CoreS2 I get a "  uint8_t port = VSPI;" error 

### Expected behavior

no error 

### Environment

- OS:
- IDE &IDE Version: Arduino 2
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-06-24:
After I get the hardware, I will conduct the test.

lboue, 2024-12-12:
@Tinyu-Zhao Did you test it?