m5stack/M5CoreS3 #59: PoweModeSet

d4rkdpg, 2024-05-10:
### Describe the bug

Hi can you please tell me how to call powerModeSet, I have a DMX module and when I used 12v in the display wont turn on, keeps power cycling. thanks

### To reproduce

. 

### Expected behavior

to work with DMX module with 12v input. 

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

d4rkdpg, 2024-05-13:
any update on this?

d4rkdpg, 2024-05-16:
can someone assist with this?

imliubo, 2024-05-20:
Hi,
Sorry for the repley late, please check [this](https://github.com/m5stack/M5Module-DMX512/issues/4#issuecomment-2119255595).

d4rkdpg, 2024-05-20:
HI i tried that, its working, although Im using CoreS3.begin(); specifically, do you have something for this also?

imliubo, 2024-05-20:
actually, CoreS3.begin() is call M5.begin() in the internal, so use M5.begin() is same.

d4rkdpg, 2024-05-20:
ok thanks

closed 2024-05-20 by d4rkdpg