m5stack/M5CoreS3 #61: M5CoreS3 & ZigBee Module

dangfan, 2024-07-12:
### Describe the bug

Are there any examples of the Zigbee Unit (SKU:U110) working with the M5CoreS3?

### To reproduce

Compiling https://github.com/m5stack/M5Stack/tree/master/examples/Unit/Zigbee_CC2630/P2P_TEST, I got an error
`fatal error: M5Stack.h: No such file or directory`
I also tried to change `M5Stack.h` to `M5CoreS3.h`, I got a new error
`error: no matching function for call to 'byteArray::printfArray<HardwareSerial>(HWCDC*)'`

### Expected behavior

An example can be provided.

### Environment

- IDE &IDE Version: Arduino 2
- Repository Version: https://github.com/m5stack/M5Stack/tree/master/examples/Unit/Zigbee_CC2630/P2P_TEST

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-07-15:
I will check it.

cslyc, 2024-12-12:
Is there an example for M5CoreS3 + ZigBee CC 2630?

lboue, 2024-12-12:
> Is there an example for M5CoreS3 + ZigBee CC 2630?

An example is needed for both cases:
- [COM.Zigbee module (CC2630F128) (SKU: M031-Z)](https://shop.m5stack.com/products/com-zigbee-module-cc2630f128-with-antenna)
- [Zigbee Unit (CC2630F128) (SKU: U110)](https://shop.m5stack.com/products/zigbee-unit-cc2630f128-with-antenna?variant=39527300137132)